analogdevicesinc/scopy #2: Default Trigger Settings

jonpearson, 2016-10-13:
As it now stands the initial Trigger state does not produce stable triggering.  Adding about 100 mV of hysteresis produces stable triggering.  The initial Trigger state should produce stable triggering for the majority of input signals so the user does not have to access the advanced triggering functions when first setting up the scope for basic operation.

dNechita, 2016-10-17:
Marking this as an enhancement for the Oscilloscope instrument.
I believe there are some techniques that can reduce the instability. We should discuss about how we can solve this issue.

adisuciu, 2017-11-07:
Oscilloscope hysteresis triggering was implemented with commit f1264ecb05139217f5ed21a557b2cda6fc28789d

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the enchancement is implemented.

closed 2017-11-21 by amiclaus