analogdevicesinc/scopy #3: Data Export

jonpearson, 2016-10-13:
Capability to export data should be added if not currently available.

dNechita, 2016-10-17:
This is on our To-Do list. The data exporting feature will be available for different formats. We should discuss which formats are a must-have.

damercer, 2016-10-17:
The obvious first format to support is comma ( or tab ) separated values, .csv. That can be read by most software like spreadsheets and MatLab ect.

Another format might be raw audio like 16 bit .wav files. If the student captures a signal they might want to listen to it or use other audio software to manipulate the data.

AlexandraTrifan, 2017-06-27:
The export feature was added for the logic analyzer with commit 88e181ad645aaa38139852bba95965cd5f6b19c0.

DanielGuramulta, 2017-08-10:
The export feature for the oscilloscope was added with commit: https://github.com/analogdevicesinc/scopy/commit/45d9464686424a4957113fc5ab46c481f45f06a0

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus