analogdevicesinc/scopy #4: Signal Generator Issues

damercer, 2016-10-14:
This is a rather long list of items and they are all included here as a time savings for me. Someone may wish to separate some of them into individual issues. There may well be more coming but this is what I've uncovered so far.

In no particular order these fall into three general categories. First implemented things that don't work properly, second changes or enhancement to already implemented things, and third things that are not yet implemented that are need ( if they are not on the "to-do list"). 

Signal Generator:
1. When you open the signal generator controls, the screen re-sizes and the bottom where you select channel 1 or 2 goes below the Task Bar and the icons can’t be seen or easily clicked on.

2. The horizontal grid scale should automatically adjust as the frequency is changed so as to show one ( or maybe two ) cycles of the waveform.

3. The vertical grid scale should automatically adjust as the amplitude and offset is changed so as to fit the waveform.

4. For the sine wave shape the amplitude setting adjusts both + and – peak values however, for all the rest of the wave shapes the amplitude setting adjusts only the + peak value. All wave shapes should work the same.

5. Scrolling the mouse wheel while hovering over the horizontal scale does something weird. Hope this is not intentional. Not sure why you need this anyway. Proper operation would be to have time “zero” always be the left edge of the grid as the scale is changed. ( or optionally have time “zero” be the center.)

6. Under Frequency you can choose mHz ( mili-Hertz? ), Hz and KHz. Where’s the MHz ( Mega-Hertz )? Not sure you need mili-Hz  but you definitely need Mega-Hz.

7. Phase ( Delay ) should be selectable in either degrees ( phase ) or time in uS / mS ( delay ).

8. For the Square wave shape where’s the duty-cycle ( symmetry ) adjuatment?

9. For the Tri-wave shape, where is the symmetry adjustment? You don’t need the sawtooth wave shape if you implement symmetry for the tri-wave. 0% would be a sawtooth one way ( up? ) and 100% would be a sawtooth the other ( down? ).

10. Sawtooth wave shape is only rising ramp? Where’s the falling ramp ( see note #9 )?

11. Should there be a built-in stair-step function? With the number of steps adjustable?

12. How do I set the rise / fall time of the square wave ( trapezoidal wave shape)?

13. Should there be a built-in noise wave shape? If so what noise distribution, uniform, Gaussian, etc?

14. There seems to be enough blank space in the configuration section ( for all but Math ) to have controls for both channels up at the same time? Saves the time clicking to switching between the two sets of controls.

15. Need to add instructions on how to use the Math controls to the Scopy Wiki page. With examples!

16. Need a few more math functions like random, sinc ( sin(x)/x ) ….

17. What file formats are supported for importing  waveforms? .txt, .csv, .wav etc. Also there should be a way to save a waveform calculated with the math controls to a file.

18. Changing the controls when outputs are enabled only effect the output if you disable and then re-enable output. You can’t adjust the values on the fly?

19. Under the constant controls, the actual output voltage is reversed, inverted. Setting to +5V outputs -5V, setting to -5V outputs +5 V and same for all the settings in between.

20. Under the waveform controls for the square wave most amplitude / offsetting work as they should but when Amp = +5 and offset = 0 the output actually swings from 0 to -5V. Same inversion happens when Amp = +5 and offset = -5 the output actually swings form 0 to + 5V.

21. Under the waveform controls for the sine wave you can set Amp = +5 and offset = -5 and the displayed wave goes from 0 to -10 which is not a valid range for the actual output. Same for a + offset.

jonpearson, 2016-11-09:
I will be separating these issues out into individual issues so they can be individually tracked.

pcercuei, 2016-11-10:
That will make things much easier. Thanks!

ABanut, 2017-05-24:
Bug closed as all issues initially reported were split on individual issues.

closed 2017-05-24 by ABanut