analogdevicesinc/scopy #5: General Issues

damercer, 2016-10-14:
Here are a few top-line issues I have with Scopy:

1. Need to remember / restore previous window settings, size, placement, location.

2. Need to save and restore previous session settings on startup. This should be a high priority to implement because it is a huge time savings for both the developers and those debugging the software. This is extended to saving a given configuration to be reloaded at a future point. This is importing for testing the software quickly.

3. Need to add an “About” page listing software rev level, firmware, hardware rev if applicable along with any other miscellaneous information. Again this should be a #1 priority to be implemented so the user can easily determine what software is running.

4. Home page displays ADI Wiki home page. It should display the Scopy Wiki User’s Guide page. ( when it is opened to the public, hint, hint ). Also the font size is too small to read. Font size should scale when window is stretched such that Wiki text fills the page ( horizontally ).

5. If only one board is found program should automatically connect to that one board. The software does not seem to always find the board when started. Can take many tries to get the board to appear.

6. Is the X vs Y plotting going to be part of the O-scope or a separate instrument?

dNechita, 2016-10-17:
1. On To Do list.
2. On To Do list. We need to do some research on ways to implementing this (whether we use or not third party libraries, what format should the session be saved in and the architectural changes that will be required).
3. This is an excellent point. We will make this a high priority.
4. We can place any page we want. Scopy Wiki User's Guide sounds good.
I think we can handle the small fonts the same way the browser usually do: add a zoom button. Maybe @pcercuei can comment of this.
5. I can see how this could speed up things from the point where the application is launched until the instruments are ready to be used. But in some cases it will be desired to connect to a remote device, which cannot be detected automatically. I believe this will no longer be an issue when the feature at point 2 will be implemented.
6. X-Y plot will be part of the Oscilloscope instrument. The XY plot window can be undocked from the main window and moved around. Right now it is not in a ready-to-be-used state.

pcercuei, 2016-10-17:
1. I think it's a bad idea, it was more annoying with the IIO Oscilloscope than anything, and the source of bugs (like when it was previously launched on an external monitor, which is not connected anymore). It should be up to the operating system to decide where to put the window. What we could memorize, however, is if the window was maximized or not and restore this setting.

3. Easier said that done; we currently have no way to determinate the firmware/hardware revisions from Scopy, the information is not made available by the firmware.

4. The web widget on the main page is meant to display the welcome screen and user guide. It should not point to an external website but to locally stored HTML documents. No zoom button, we can use proper fonts directly in the HTML/CSS.

5. I think auto-connect makes sense. But you say 'the software does not seem to always find the board when started' and it's the first time I hear about this and it works 100% of the time here. Could you elaborate?

damercer, 2016-10-17:
I can see your point for users that might have two computer displays but I would guess that is a small sub set of users. Is there a way for the program to ask the OS if the saved window location will not be off the current display space? I think there might be. 

In general I'm apposed to just letting the OS "decide" how I want my screen arranged.

damercer, 2016-10-17:
The problem of the program not finding the board was random and not reproducible. It seemed to happen most often the first one or two times I opened Scopy and then it started finding it. It might have had something to do with starting up on a virgin installation.

damercer, 2016-10-17:
If the current hardware / firmware does not support versioning then this is something that needs to be designed in right away. We waited too long till the 11th hour to add this to the M1K and should not have.

rgetz, 2016-10-17:
Things do support versioning.

There are ways to find versions of all the components:
 - version of scopy (should be built in)
 - version of libiio on host 
 - version of libiio on backend
 - version of kernel on backend (uname -a, Paul? - I'm not sure this gets exposed)
 - version of HDL on backend (may need to be added)
 - version of hardware on backend (may need to be added).

-Robin

pcercuei, 2016-10-18:
The problem is how to add those? That's not something we can export through IIO.

dNechita, 2016-10-19:
It looks like there is need for another daemon to run on the target that has the capability to perform a couple of system interrogations and send them through USB or Ethernet. Moreover this can be used for eeprom reads/writes and for triggering updates such as  ADI tools and ADI boot image on target.

rgetz, 2016-10-19:
Or - make an iio device attribute "version", which encapsulates HDL and kernel. That would make sense to me.

pcercuei, 2016-10-20:
I'm not sure @lclausen-adi would be happy with this :)

lclausen-adi, 2016-10-20:
It is not a good idea, would not make it into upstream (neither kernel nor libiio).

amiclaus, 2017-10-11:
I'm closing this list of issues since most of them were debated/implemented and moving nr. 3 to a separate issue.

closed 2017-10-11 by amiclaus