analogdevicesinc/scopy #7: Status or control buttons

rgetz, 2016-10-15:
Was part of #6

The word in the “enable” button should change to “disable” when the supply is on as well as change color, because the function of the button has changed.

rgetz, 2016-10-15:
Yeah - this is one of the things I always struggle with. When a button combines control and status, does it display the control "turn this off", or the status "this is running". 

As long as things are consistant across all instruments - it should be fine.

What about this:

When an instrument is "off/not running", it displays a green "Run" button. Clicks to this button either put it in the cal or run state. 

When an instrument is calibrating (if necessary), it displays a yellow "Cal" button. Clicks during this time takes it to the off state. No clicks take this to the run state. 

When an instrument is "on/acquiring/running" it displays a red "Stop" button. Clicking on this button take it to the off state. 

Issue with this - is green means off, red means running. It's a little goofy from that standpoint.

damercer, 2016-10-15:
It can take up more screen space but having two buttons rather than one. One always red for stop one always green for run. Pushing stop while stopped is a nop or could be a screen refresh.

pcercuei, 2016-10-17:
The colors we've settled with for now are yellow for 'off/not running' and red/orange for 'running'.
The exception is the 'connect' button in the main page, which turns green if successfully connected, and red if the connection failed.

In the mockups, the 'Run' button is green, the 'Single' (for single-shot capture) is yellow, so I can adapt it to match better the mockups.

I don't like the idea of having two buttons - I think one is enough as long as the action is not ambiguous; 'Run' and 'Stop' captions should be OK.

damercer, 2016-10-17:
The yellow and orange colors ( as displayed on this laptop screen ) are too close in shading to be obviously different. Red and green show much better contrast, unless you are R/G colorblind of course.

pcercuei, 2016-10-17:
Yes, that's why I suggested to use green instead of the current yellow.

DanielGuramulta, 2017-05-11:
Can I close this, since the initial issue has been solved?

pcercuei, 2017-05-11:
Yep - it can be closed.

closed 2017-05-11 by pcercuei