analogdevicesinc/scopy #9: Oscilloscope Comments:

damercer, 2016-10-15:
Here are a few initial comments on the Oscilloscope tool.

1. The grid has no labels. The horizontal axis needs time labels and the vertical axis should have color coded labels for each channel ( because position and scale might be different ).

2. The minor tick marks are along the top, bottom, left and right sides of the grid. Since the waveform seldom if ever crosses the top or bottom edges the minor ticks should be along the center horizontal grid line where the wave will cross. For symmetry there should then also be minor ticks along the vertical center grid line. Placing ticks here makes it easier to use them to judge the waveform values ( based off the grid marks).

3. The little red numbers that track the cursor are clipped off on the right side so that the vertical value has no units. Can't tell, 80 mV is displayed as 80.

4. Having two set of independent trigger controls for channel 1 and channel 2 is confusing. Most scopes have one set of controls and they apply to whichever channel is selected as the source.

5. The horizontal position controls are confusing. The time "zero" grid line ( not labeled, see number 1 above ) should be able to be set to any position along the grid and the trigger point is always at time "zero". Now it seems that the trigger position is always the center grid. Sliding the horizontal position indicator ( light blue line ?) will also slide the trigger point on the grid to match ( Time = "zero" )

6. You might consider adding a "set trigger level to 50%" button where the trigger level for the currently selected trigger source is set to mid way between the min and max values of that waveform.

7. Again this tool suffers from the same resizing issue. The widow often goes off the edge of the screen ( bottom and, or right ) and you can't get it back.

8. Are there plans to have the waveform color changeable by the user? Also have the background be able to be black or white? I like the black background personally but for screen shots that you might want to include in a report document ( that gets printed on paper ) you might want a white background.

dNechita, 2016-10-17:
1. If by grid label you mean the values of each major tick of a grid then we are not going to display them. I believe this was the plan from the beginning. We can discuss this.
2. I will make a quick sketch so that we can get an idea about how it would look and we can further discuss this.
3. It is a known issue. Is this feature useful? I was thinking of removing it entirely.
4. The current hardware supports triggering on both channels simultaneously and allows some logical operations between the two trigger conditions. E.g. (T1 and !T2), (T1 or T2), etc. If we allow only one of the trigger to be active at certain time, then the logical operations feature will never be used. We should discuss this.
5. Maybe @pcercuei can comment on this.
6. I was planning to add this, but slightly different, like pressing a button that would reset the trigger level to 0 Volts. Setting this to 50% of peak-to-peak sounds even better.
7. We will try to solve it.
8. Yes, the color of the waveforms should be changeable as well as the thickness of the line. I believe that we will provide two themes for Scopy: a dark and a light one. The light theme will have a white background. Whatever the theme is, we should be able to implement a printing option that will produce a .png of the plot that has a white background.

damercer, 2016-10-17:
"we are not going to display them" I could not disagree more!!! The the grid absolutely needs to have values next to the major gird lines. This should not be negotiable. Also it might be a good idea to have the vertical and horizontal scales i.e. v/div uS/div placed somewhere on the main display.

damercer, 2016-10-17:
Google "oscilloscope graticule" and you will see what I mean:

https://www.google.com/search?q=oscilloscope+graticule&biw=1173&bih=605&tbm=isch&tbo=u&source=univ&sa=X&sqi=2&ved=0ahUKEwjjudryg-LPAhXF2SYKHU73Cv8QsAQIZQ

Oscilloscope screens have always had the minor tick marks along the center horizontal / vertical major grid lines.

damercer, 2016-10-17:
I'm not sure I follow what you are trying to do having simultaneous triggering on both channels. Would that mean that if I set the triggering for both CH1 and CH2 on a rising edge but the two signals are of the same frequency but 180 out of phase they will be displayed ( wrongly ) with there rising edges aligned at time "zero"?

I hope you are not considering sliding the waveforms so they align their respective trigger events at time = 0.

Now if you are considering having what is called "complex" triggering where logical conditions between events on the two channels then that is another matter. Like trigger only on CH1 rising edges that happen after a rising edge on CH2 etc.

rgetz, 2016-10-17:
Doug:

While I agree we need to display the scale (V/div for each channel and s/div) somewhere, I don't think we need the numbers by the grid. When we surveyed most bench instruments - they don't do this. (Even your link shows nearly all don't have this).

Agree on the graticule comment - that was an oversight in the mockups, we should update things to the centre.

There are many scopes which have complex triggers - it could be a question of the default view, and what gets exposed to the user to start.

As for re-sizing issues - what is your screen resolution set to, and what is the smallest you think we should support? I think we were hoping to 1024 x 768 would be the smallest....

This is the issue with posting multiple items in one thread - the discussion gets intermingled, and is difficult to follow.

dNechita, 2016-10-17:
I will start splitting things in separate issues.

damercer, 2016-10-17:
Robin:
For analog scopes it was impossible to label the axis or even add V and H scale factors until Barrie Gilbert invented how to do it. As for DSOs it is probably a hold over from the past. Just because it was not done in the past is not sufficient reason not to do it now. Digilent Waveforms labels the axis. Pixelpulse2 does it in a fashion. Spectrum / Network Analyzers have always had axis labels. Some PC scope software ( like for sound cards ) have time axis labels at least ( sound card vertical is un-calibrated so labels would be relative at best).

damercer, 2016-10-17:
As for triggering I think you should hide the simultaneous ( complex ) triggering features until the user opens them up in a special menu. Too confusing when all the user wants is to do is trigger on a waveform to get a steady display.

damercer, 2016-10-17:
The lap-top screen is 1600 X 900. But the software should be able to query the screen dimensions and never re-size the window to draw outside the Box as it were. Better to just fit the stuff into the current window size whatever it might be.

dNechita, 2017-05-03:
Closing this as all issues have been split into new issues.

closed 2017-05-03 by dNechita