analogdevicesinc/scopy #10: Color underlines

damercer, 2016-10-17:
I have an observation about the orange underlines that are used in various places like indicating which channel the controls are operating on. The use of the same color, orange, as the waveform color for CH1, is confusing under the channel 2 heading label. Use the CH2 color under channel 2 heading or always use some other color not related to the channels.

DanielGuramulta, 2017-07-28:
I've changed the color for the tabs, it looks like this: 
[image].
[image]
Here is the branch with the changes: https://github.com/analogdevicesinc/scopy/tree/siggen_quickfix

DanielGuramulta, 2017-08-24:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/6fd7ece53a1010c2eb7ed401e4067028c5c0163f

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus