analogdevicesinc/scopy #12: Oscilloscope: A graticule is required

dNechita, 2016-10-24:
Moving this from a list of issues at #9 into its own issue.

Initial issue:
"The minor tick marks are along the top, bottom, left and right sides of the grid. Since the waveform seldom if ever crosses the top or bottom edges the minor ticks should be along the center horizontal grid line where the wave will cross. For symmetry there should then also be minor ticks along the vertical center grid line. Placing ticks here makes it easier to use them to judge the waveform values ( based off the grid marks)."

dNechita, 2017-05-24:
Will be adding this as an option in the settings of the oscilloscope instrument.

claudiupop, 2018-06-04:
Implemented with https://github.com/analogdevicesinc/scopy/pull/441

amiclaus, 2018-07-31:
On Scopy build: 01d69df (v1.0.1) the feature is implemented.

closed 2018-07-31 by amiclaus