analogdevicesinc/scopy #16: Oscilloscope: A set-trigger-level-to-50% button would be helpful

dNechita, 2016-10-24:
Moving this from a list of issues at #9 into its own issue.

Initial issue:
"You might consider adding a "set trigger level to 50%" button where the trigger level for the currently selected trigger source is set to mid way between the min and max values of that waveform."

rgetz, 2016-10-25:
Do we need a 50% button, or an autoset button?

I think autoset would get more use...

-Robin

damercer, 2016-10-26:
Sometimes ( most times ) an "Auto" set button does more than just adjusting the trigger level so it probably will alter setting that you wanted left untouched. Both kinds would be useful.

amiclaus, 2018-05-21:
Yes, and also with #161

amiclaus, 2018-08-01:
On Scopy build: 01d69df (v1.0.1) the feature is implemented.

closed 2018-08-01 by amiclaus