analogdevicesinc/scopy #18: Oscilloscope: User selectable waveform and background colors

dNechita, 2016-10-24:
Moving this from a list of issues at #9 into its own issue.

Initial issue:
"Are there plans to have the waveform color changeable by the user? Also have the background be able to be black or white? I like the black background personally but for screen shots that you might want to include in a report document ( that gets printed on paper ) you might want a white background."

dNechita, 2017-05-24:
Need mokcup for the color chooser widget.

pcercuei, 2017-05-24:
About the background color - maybe that's something the theme should provide?
Then we can have a dark theme (the current one) and a light theme.

dNechita, 2017-05-24:
Yes, the background color should be provided by the theme.
In case we need to export a .png of the plot, we should add options like "Black and white" or "Invert colors" that would make the background white in case it is needed for printing.

rgetz, 2018-04-26:
+1 for comment about printing with white background. I think we are running into this as people make up their labs.

-Robin

rgetz, 2020-10-22:
for managing preferences, I think you just have a list of features and a choose able colour by each. Something similar to (but not exactly); I still think we should keep a "screen" theme, and a "print" theme (which might be dark/light by default).
[image]

for picking colours, I think we would just use [QColorDialog](https://doc.qt.io/qt-5/qcolordialog.html) (or similar)
[image]

adisuciu, 2020-10-22:
We could do that, we just need to figure out what our features are. 

Another quick and dirty way to do it is with stylesheet files. Most of the stylesheets are part of a single file(global.qss) already. We could load similar files at runtime and have the file bundled with the executable as failsafe. This way you can create your own themes - or have a selection inbetween multiple stylesheet files that are provided with Scopy - similar to what we did for languages. It's less user friendly but more customizable.

rgetz, 2020-10-22:
it's a tradeoff between making it customizable, vs how difficult it is to customize...

I don't want to be in the business of making customizations for everyone, because it's too hard for users. (as I'm sure you don't either)...

connorka, 2020-10-22:
In addition to printing, slides are best done with light colors, which is also generally true for videos. I agree with Robin that one does not need a lot of options, at least for backgrounds. For signals/line plots etc, having the ability to choose colors, as one does with Excel and Matlab, can be very useful.

rgetz, 2020-10-26:
Also - this is more than just scope, but the all instruments, should have a coordinated colour scheme.

I was trying to find an equivalent (scope or spectrum analyzer) settings - and I couldn't find any in tek or R&S, or other traditional instrumentation manufactures, but did find:
- [BitScope](https://docs.bitscope.com/KLTUKBXN/#eabum3sbcw0x)
- [Waveforms main options](https://reference.digilentinc.com/reference/software/waveforms/waveforms-3/reference-manual#options) have a "dark" and "light". then you change the colour of the traces on each instrument - [Waveforms Scope](https://reference.digilentinc.com/reference/software/waveforms/waveforms-3/reference-manual#real_channels) or [Waveforms Spectrum Analyzer](https://reference.digilentinc.com/reference/software/waveforms/waveforms-3/reference-manual#trace)
- [Picoscope](https://www.picoauto.com/support/viewtopic.php?t=17751)
- [Hantek](http://www.hantek.com/Product/Hantek6000/HT6082_Manual.pdf#page=65)

and a few other USB scopes...