analogdevicesinc/scopy #20: Oscilloscope and Signal Generator

wkester, 2016-11-04:
Clicking around on the vertical or horizontal axis sometimes makes the waveform disappear. 
Would it make sense to put in a "reset" button to restore to the default settings?

rgetz, 2017-03-13:
I see this on the signal generator.

double click in the graph would be a convenient way to autoscale.

[image]

I do get this quite often:

[image]

If you want to plot out to 5 seconds - that's fine, but then there should be data...

rgetz, 2017-03-13:
You should see at least two periods.

dNechita, 2017-05-28:
Fixed in https://github.com/analogdevicesinc/scopy/commit/d42f260809a169998dd5fb0cc0fe78702caff237
A right click brings you to the default zoom (acts as a reset).

rgetz, 2017-05-28:
How do you right click on a touch screen?

Does pinch work?

dNechita, 2017-05-29:
The right click functionality is something that came with the plot widget that we're using. We didn't start adapting this functionality (or others) to work on touch screens yet.

adisuciu, 2017-10-20:
The zooming/preview system was reworked with f2876e55e1d1ece4a1b2f3d358a623d9a0368e55 and f3e3f436ce07e6fc18038d2370f5f0a8163951df . The signal will always display 2 periods of the slowest waveform.

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus