analogdevicesinc/scopy #21: Item 1 From Issue #4: Screen Resize When Opening Signal Generator

jonpearson, 2016-11-09:
Item 1 from Issue #4

"When you open the signal generator controls, the screen re-sizes and the bottom where you select channel 1 or 2 goes below the Task Bar and the icons can’t be seen or easily clicked on."

jonpearson, 2016-11-09:
I was not able to reproduce this issue in the current revision of Scopy.  The channel selection options were above the Task Bar when I opened the signal generator.

rgetz, 2016-11-10:
This will depend on the screen size. 

I think Doug's laptops are 720p displays. (1280×720 px). If you are using 1080 - you will not be able to reproduce. 

Change the resolution on your screen to 720p, and try it out.

damercer, 2016-11-10:
Correct. The software needs to know the screen size so it does not grow the window beyond the visible space. In my opinion the program should not change the window size beyond what
the user has set. And remember that for future sessions.

pcercuei, 2016-11-23:
I'll mark this as duplicate of #17 and close this one.

closed 2016-11-23 by pcercuei