analogdevicesinc/scopy #22: Item 2 From Issue #4:

jonpearson, 2016-11-09:
Item 2 from Issue #4 

"The horizontal grid scale should automatically adjust as the frequency is changed so as to show one ( or maybe two ) cycles of the waveform."

jonpearson, 2016-11-09:
This would be a good enhancement, but is not a super high priority because many signal generators do not display their output waveforms.  As is stands, it's mostly useful as an indication of signal amplitude and offset.

damercer, 2016-11-10:
Displaying the waveform is a nice idea but as was said not something that most generators do. It takes up a lot of screen space and as Scopy is currently configured you display only one set of instrument controls at a time. It is painful to have to switch back and forth between AWG screen and scope screen all the time. Having a minimalist set of AWG controls displayed along side the scope controls would be a huge time saver.

adisuciu, 2017-10-20:
Two periods of the signal is displayed with the latest signal generator preview rework. Implemented in commit f2876e55e1d1ece4a1b2f3d358a623d9a0368e55 . It is now possible to have each instrument in it's own window, so one can easily change the waveform from the signal generator while looking at the oscilloscope.

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the enchancement is implemented.

closed 2017-11-21 by amiclaus