analogdevicesinc/scopy #24: Item 4 From Issue #4:  Adjust All Waveform Amplitudes in the Same Manner

jonpearson, 2016-11-09:
Item 4 from Issue #4 

"For the sine wave shape the amplitude setting adjusts both + and – peak values however, for all the rest of the wave shapes the amplitude setting adjusts only the + peak value. All wave shapes should work the same."

jonpearson, 2016-11-09:
The sine waveforms initializes as a bipolar waveform and the others are positive-only.  It would be good to have all waveforms come up bipolar and have the amplitude adjustments symmetrically change the peak-to-peak value.  The knob setting should indicate the waveform amplitude.

rgetz, 2016-11-10:
Is the amplitude knob meant as a peak-to-peak amplitude (0-10V) or a single ended amplitude (-5V - +5V)?

I would expect that it would always be peak-peak?

damercer, 2016-11-10:
In Digilent Waveforms the amplitude is peak not peak to peak. If you set the amplitude to 2 V the sine wave goes from -2 V to +2 V. As it would in the formula Amp * Sin(wt) + Offset.

rgetz, 2016-11-10:
Doug: I agree - I always thought Waveforms was broken/goofy. 

I would rather compare to Agilent:
http://literature.cdn.keysight.com/litweb/pdf/33500-90901.pdf?id=2197440

When adjusting amplitude - units are either Vpp, or Vrms, or dBm (page 39 of the doc). 

? We can do it either way - which is the "standard"

damercer, 2016-11-10:
The thing is as you query, What is "Standard" practice.? I go back to Digilent Waveforms all the time because our user base is likely to be current ( then former ) Analog Discovery users. And following suit with that ( rightly or wrongly ) would be least confusing.

commit f373d788 (2016-11-21): Signal Generator: Adjust all waveforms amplitudes in the same manner

Fixes #24.
closed 2016-11-21 by pcercuei