analogdevicesinc/scopy #25: Item 5 From Issue #4:  Behavior When Mouse Wheel Scrolled Over Horizontal Scale

jonpearson, 2016-11-09:
Item 5 from Issue #4 

"Scrolling the mouse wheel while hovering over the horizontal scale does something weird. Hope this is not intentional. Not sure why you need this anyway. Proper operation would be to have time “zero” always be the left edge of the grid as the scale is changed. ( or optionally have time “zero” be the center.)"

jonpearson, 2016-11-09:
Doing this caused the number of cycles displayed on the screen to change with a non-corresponding change in the timebase.  The numerical value of frequency in the control area did not change.

adisuciu, 2017-10-20:
In my opinion we should remove scrolling over axes, since scrolling is not touch friendly.

DanielGuramulta, 2018-01-05:
The mouse gestures are no longer enabled by default since commit: https://github.com/analogdevicesinc/scopy/commit/69a1b689800f6cbe60c66823cc4c05c05ecd25cb

amiclaus, 2018-01-16:
On Scopy build e9a6c85 (v0.10) the bug is fixed.

closed 2018-01-16 by amiclaus