analogdevicesinc/scopy #26: Item 6 From Issue #4: Megahertz vs. Millihertz and How to Distinguish Between Them

jonpearson, 2016-11-09:
Item 6 from Issue #4 

"Under Frequency you can choose mHz ( mili-Hertz? ), Hz and KHz. Where’s the MHz ( Mega-Hertz )? Not sure you need mili-Hz but you definitely need Mega-Hz."

jonpearson, 2016-11-09:
Using a lower case "m" for "milli" and an upper case "M" for "Mega" would work, but I don't think we need millihertz.  Adding a "MHz" option to the pulldown menu below "kHz" and eliminating the "mHz" option would be a good solution.

pcercuei, 2016-11-10:
millihertz was a requirement from @rgetz, since we might want to generate very slow waveforms.
Megahertz is not possible yet, since the signal generator only runs at 750 kHz.

rgetz, 2016-11-10:
Paul:

The signal generators run at 100MHz. It should be possible to get a reasonable 10MHz sine wave out. And easy to get a 20MHz square wave.

pcercuei, 2016-11-10:
No, the signal generator runs at 750 kHz. The hardware can do 75 MHz, Scopy configures it to 1MHz and gets 750 kHz (so with a /100 divisor). We can't do higher until we have one DMA per channel which would allow us to have cyclic buffers.

rgetz, 2016-11-10:
Right - and we said that wasn't going to be for Alpha, but needs to be there before Beta. What is the schedule? 

I thought @acostina said the cyclic DMA was ready. (but could have been mistaken). Adrian - can you confirm?

Thanks

acostina, 2016-11-17:
The latest FPGA project has one DMA for each TX channel of the AD9963, with the cyclic option active. The interpolation module supports different interpolation rates per channel. 

Not sure if the linux driver supports this infrastructure at this time.

pcercuei, 2017-05-24:
I think this works fine now.

closed 2017-05-24 by pcercuei