analogdevicesinc/scopy #28: Item 8 From Issue #4:  Duty Cycle Adjustment for Rectangular Waveform

jonpearson, 2016-11-09:
Item 8 from Issue #4 

"For the Square wave shape where’s the duty-cycle ( symmetry ) adjuatment?"

jonpearson, 2016-11-09:
If we limit the waveform to square, then by definition the duty cycle is 50% and should not be adjustable.  For the most flexibility, we should relabel the waveform as "Rectangular" and provide a duty cycle adjustment.

damercer, 2016-11-10:
Digilent Waveforms has the Symmetry knob which for a square wave is duty cycle.

I think most people include any flat top waveform with "vertical" edges as Square Waves.

Waveforms with slower edges are referred to as Trapezoidal waveforms.

adisuciu, 2018-01-12:
Implemented with b267d3c4d8a72e7b615fbbe78ce66963fe9dfcb1

amiclaus, 2018-01-16:
On Scopy build e9a6c85 (v0.10) the feature is implemented.

closed 2018-01-16 by amiclaus