analogdevicesinc/scopy #32: Item 12 From Issue #4

jonpearson, 2016-11-10:
Item 12 from Issue #4 

"How do I set the rise / fall time of the square wave ( trapezoidal wave shape)?"

jonpearson, 2016-11-10:
This feature needs to be added.  Should we change the name to "Rectangular?"  Doug's indication of "trapezoidal" is the most accurate, but it's not used on traditional function generators.

rgetz, 2016-11-10:
http://literature.cdn.keysight.com/litweb/pdf/33500-90901.pdf?id=2197440

Square waves : frequency, amplitude, offset, duty cycle, and phase

Pulse : : period, amplitude, offset, width, leading edge, trailing edge, and phase.
The edge time represents the time from the 10% threshold to the 90% threshold of each edge.

So, the function is there - it's just not called "trapezoidal", it's a pulse.

I think for what we are doing - trapezoidal is actually clearer (IHMO).

damercer, 2016-11-10:
Everybody seems to have their own preferred names for various wave shapes. My take it that the name is less important than the functionality to control all the various aspects of the wave shape.

adisuciu, 2018-01-12:
Commit b267d3c4d8a72e7b615fbbe78ce66963fe9dfcb1 implements trapezoidal waveform that allows setting rise/fall times for rectangular waveforms.

amiclaus, 2018-01-16:
On Scopy build e9a6c85 (v0.10) the feature is implemented.

closed 2018-01-16 by amiclaus