analogdevicesinc/scopy #36: Item 16 From Issue #4:  Add More Math Functions

jonpearson, 2016-11-10:
Item 16 from Issue #4 

"Need a few more math functions like random, sinc ( sin(x)/x ) …."

jonpearson, 2016-11-10:
This would be good for a future enhancement.

rgetz, 2016-11-10:
sinc can be done (sin(x)/x) - providing primitives is what is required - isn't it?

damercer, 2016-11-10:
This may be true but the sinc function as implemented by numerical function libraries ( numpy for example ) make it simple and easy to do. They include more parameters than just X.

rgetz, 2016-11-10:
Doug:

I agree - but it's a question of where to stop? If you want numpy inside - that might be easier than trying to add every math function possible...

http://literature.cdn.keysight.com/litweb/pdf/33500-90901.pdf?id=2197440

They take a standard waveform, and then apply various "advanced math" pieces to it:
absolute function, invert, mirror (reverse the waveform), scale, clip, trim, filters, sequencing (playing X snippet A times, then Z snippet for B times), Quasi-Gaussian Noise, A Pseudo-Random Bit Sequence (PRBS), AM, FM, FSK, BPSK, PWM, Sum, Burst, Three-Cycle Burst, Frequency Sweep, - are but a few different "math" functions that can be done.

Some of those are in WaveForms, but I'm not sure we know which are actually used by students/faculity.

If you can help be more specific of which you are looking for (since you wrote so many labs) we can look at those, and try to both rank them in order of priority - then we can respond back better with a schedule, and what we think we can do before Dec.

Thanks
-Robin

damercer, 2016-11-11:
You are right there are possibly a huge number of useful signal processing functions that might be included. A lot of the waveform generation software provided by AWG instrument makers include about 6 or 8 of the more common wave functions.  But for me in writing ALICE it was easiest to include access to all of the numpy library because it was already being used for other things like doing FFTs in the SA, NA, and IA etc. Providing User access to the functions is trivial with the Python interpreter.

In Scopy are these sorts of math functions being re-written from scratch as needed or are you including a pre-written open source library of "numerical" functions?

With Scopy being written in C++ I'm guessing it would be harder to build in an interpreter that can parse equations and send to a numerical library. I don't know, that might not be that hard....

The up side of doing this extra level of work is that you can also use this on the scope side to plot signal processing functions applied to measured waveforms as well. Which is where I use the math library the most in the Labs I've developed. Calculating power and energy, taking derivatives, plotting histograms, digital filtering, a whole list of  equations.