analogdevicesinc/scopy #38: Item 18 From Issue #4:  Changing Settings on the Fly Without Having to Disable/Enable

jonpearson, 2016-11-10:
Item 18 from Issue #4 

"Changing the controls when outputs are enabled only effect the output if you disable and then re-enable output. You can’t adjust the values on the fly?"

jonpearson, 2016-11-10:
This was somewhat of a nuisance to me as well, and is not required on typical benchtop signal generators, but may be more common on more virtual-type instruments.  I think it would be good to be able to change waveforms on the fly if it is feasible.

rgetz, 2016-11-10:
This may be more of a backend issue than anything -- it's not possible to change streams/configurations on the fly, 

but it should be possible to silently stop things, change a setting, and then start things again, without the user having to do those setting manually - so it looks like things are changing in real time.

-Robin

damercer, 2016-11-10:
To the user it does not really matter what is going on under the hood. They just want to quickly make the adjustments they want as easily and quickly as they can. A slight burp as the change happens is often to be expected.

commit 95cdfb61 (2017-03-02): Signal Generator: Allow the settings to be changed at runtime

Fixes #38.