analogdevicesinc/scopy #39: Item 19 From Issue #4:  Function Generator Constant Output Voltage Inverted

jonpearson, 2016-11-10:
Item 19 from Issue #4 

"Under the constant controls, the actual output voltage is reversed, inverted. Setting to +5V outputs -5V, setting to -5V outputs +5 V and same for all the settings in between."

jonpearson, 2016-11-10:
I could not duplicate this.  The DC levels displayed on the oscilloscope were of the correct polarity, and were verified with a DMM.  There was an offset in the function generator output level.

damercer, 2016-11-10:
This inversion only happens at the + and - extremes of 5 V as with the waveforms. Setting the DC value to +5.00 outputs -5 but +4.9 outputs ~ +4.9. So I think cause of this is the same as the what we see when the sine amplitude is set to the Max. 5.00

( may have miss spoke with the " same for all the settings in between" part of the comment. )

pcercuei, 2016-11-22:
I'll close this one, as it's a side effect of #41.

closed 2016-11-22 by pcercuei