analogdevicesinc/scopy #40: Item 20 From Issue #4:  Square Wave Swing Incorrect for Large Amplitudes

jonpearson, 2016-11-14:
Item 20 from Issue #4 

"Under the waveform controls for the square wave most amplitude / offsetting work as they should but when Amp = +5 and offset = 0 the output actually swings from 0 to -5V. Same inversion happens when Amp = +5 and offset = -5 the output actually swings form 0 to + 5V."

jonpearson, 2016-11-14:
These were verified with an oscilloscope, with details below.

For the square wave set up to swing positively, the following anomalies occurred at the given amplitude settings with offset = 0:

Amplitude = 4.28 V     Signal looked normal
Amplitude = 4.29 V     Negative-going glitches began to appear and got wider as amplitude increased

Amplitude = 4.66 V     Signal was still in the positive region with negative-going glitches
Amplitude = 4.67 V     Signal shifted to swing in the negative region (from about 0 V down to the negative of the amplitude setting) with positive-going glitches

For the square wave set up to swing negatively, the following anomalies occurred at the given amplitude and offset settings:

Amplitude = 4.28 V, Offset = -4.28 V     Signal looks normal
Amplitude = 4.29 V, Offset = -4.29 V     Positive-going glitches began to appear and got wider as amplitude and negative offset increased

Amplitude = 4.66 V, Offset = -4.66 V     Signal was still in the negative region with positive-going glitches
Amplitude = 4.67 V, Offset = -4.67 V     Signal shifted to swing in the positive region (from about 0 V up to the amplitude setting) with negative-going glitches

The anomalies occurred at exactly the same amplitude settings in each region.

amiclaus, 2017-10-12:
From my point of view, after some checks, this issue is outdated and invalid for the latest release 290a894 (v0.8). Therefore, I will close it.

closed 2017-10-12 by amiclaus