analogdevicesinc/scopy #42: Oscilloscope Input Coupling

jonpearson, 2016-12-02:
It would be good to have an option to have each channel of the scope and possibly the external triggers AC- or DC-coupled.  This can be viewed as an enhancement.

damercer, 2016-12-10:
This will require hardware changes to include switches directly at the analog input connector pins. These pins can go beyond the +/- 5V supplies and would thus require actual relays not analog switches.

rgetz, 2016-12-10:
Doug: it depends on if by "AC" it means it's true AC coupled in hardware (like you indicated), or if the DC component is removed before display .... (totally lieing to the end user, and faking it in software). 

A high pass filter is pretty easy (or just find the average/median, and subtract it before displaying it). 

It really depends on the experiment that Jon/you are thinking about.

damercer, 2016-12-11:
Robin: That is an important distinction. If the aim is to center the waveform on the screen then that is what  the vertical position controls are for. If in addition the idea is to constantly adjust the position to keep the waveform centered as the DC average changes that is a very useful feature. In ALICE it is called "Auto-Center". Maybe not completely descriptive but in any case such a software feature should be called something other than AC coupling to avoid confusing or misleading the user. ( There are USB scopes from SYSCOMP and Velleman that have real AC coupling. )

As for "AC coupling" on the Trigger inputs. Again there is this hardware vs software distinction. The M2K has the two digital I/O pins that can be triggers and I'm not sure how software AC coupling would work there. Having the trigger level on the scope analog inputs automatically track the center of the waveform is again a useful feature ( called Auto-Level under triggering in ALICE  ).

rgetz, 2016-12-11:
Doug:

Yes - there are a lot of differences between the methods and results of different implementations of "AC coupling". Like you mentioned:
 - actual AC coupling with a cap.  I think we were thinking about something similar to (or just pointing people to):
http://store.digilentinc.com/bnc-adapter-board-for-the-analog-discovery/
This has jumper to select AC or DC coupled. (So it's not automatic, has no GUI, but it is cheap)
 - high pass digital filter. Like you alluded to - the issue is always dynamic range. If you have a 5mV signal @ 10V, it's hard (impossible) to do this with a 12-bit converter. At +/- 10V one LSB is 4.8mV, so nominally - the signal gets lost in the noise. It works OK for larger signals. It all depends on the experiment. 
 - vertical offset. On devices like M2k, whichbhave vertical offset adjustment in the analog domain, this is solves the dynamic range issue - but the settings will never be  real time - it will always be servoing to the average of the last buffer - which would cause the actual signal to bounce up and down - unless you also used a digital filter, but then the implementation gets super complex - very quickly. 

Again - what the "best" implementation is depends on the goal, - and what experiments are being done. 

Maybe Jon could add what we was thinking when he added this. 

- Robin

damercer, 2016-12-12:
One more quick observation. It should not be too complicated to combine the analog hardware offset of the M2K to remove the majority of the DC ( which should get you close enough to use the higher gain settings ) with software subtraction of the remaining DC average of the current buffer. That should remove any bouncing from sweep to sweep.

AlexandraTrifan, 2017-12-15:
AC coupling was added in software and should be available after commit d38ef366e0bf46628ebda35b32e0441ffdf17faf.

amiclaus, 2018-01-16:
On Scopy build e9a6c85 (v0.10) the feature is implemented.

closed 2018-01-16 by amiclaus