analogdevicesinc/scopy #46: Function Generator Output Overload, Oscilloscope Amplitude Accuracy and Offset

jonpearson, 2016-12-02:
When the function generator output is set to 5 Vpeak (initial condition), the output exhibits phase reversal distortion -- may be related to issue #40.  For function generator amplitude settings of 4.66 Vpeak, the waveform shapes observed on the Scopy oscilloscope and a lab oscilloscope look okay.  The amplitude of the waveform on the Scopy oscilloscope is too large, however, and is clipped on the display.  The function generator output overload condition begins to manifest itself for amplitudes greater than or equal to 4.67 Vpeak.  There is also an offset that appears to be in the Scopy oscilloscope.

pcercuei, 2017-05-25:
I think this is fixed.

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus