analogdevicesinc/scopy #47: Scopy Windows Installer

damercer, 2016-12-10:
The  Windows installer for Scopy forces a shutdown / restart without warning the user that it is about to happen. This is bad form. You need to clearly state that a re-start is needed before Scopy can be run but ask the user if it is OK to do it now or wait for them to save / close any apps that might be running first.

pcercuei, 2016-12-12:
The installer clearly warns about the reboot and even gives you the option to perform the reboot yourself. See attached screen capture.

[image]

You don't have that?

damercer, 2016-12-12:
Not on my ADI issued Windows 7 laptop it doesn't. I tried twice. Otherwise why would I have posted the issue? Your response sounded like I just didn't see it or something.

pcercuei, 2016-12-12:
So what installer are you using?

rgetz, 2017-03-13:
I just tried the Alpha-2, and it shows things on my laptop.

-Robin

closed 2017-03-13 by rgetz