analogdevicesinc/scopy #48: Scopy Using FIR filters in AD9963

damercer, 2016-12-15:
The AD9963 has built in FIR filters in the Tx path. It looks like these are being used i.e. not being by-passed in present Scopy configuration. This leads to the funny shape of a square wave as seen here:
[image]

The pre-shoot before the edges comes from the impulse response of the filter.

Also I'm not entirely sure of this theory but having the filter(s) in the signal path might be the cause of the apparent phase reversal at the top and bottom of the signal range. Here is a scope shot of the same square wave at a slightly higher amplitude where the tips if the pre shoot goes over the magic amplitude value and reverses wraps around:
[image]

In either case the internal filters of the AD9963 signal paths should be by-passed by default. Optionally allowing the user to use the filters could be useful but only for sinusoidal signals if you are looking at them in the time domain.

Doug

dNechita, 2017-05-25:
@amiclaus can you please try to reproduce?

damercer, 2017-05-25:
This may have gone away at some point. I've not seen this problem to this extent for some time. It might have happened when the sample rate was increased to the full 100 MSPS for the AWG.

amiclaus, 2017-10-12:
This issue is fixed on 290a894 (v0.8)

closed 2017-10-12 by amiclaus