analogdevicesinc/scopy #50: Hardware support for delayed aquisitions is needed

dNechita, 2017-01-10:
About:
One of many oscilloscope features is to provide the option to set the trigger position (in time). The position can be negative or positive as the starting point (position 0s) is at the middle of the screen.

Request:
Hardware support for delaying an acquisition that would otherwise start immediately, once a trigger condition has been met, with a given amount of time.

Request reason:
With the current hardware capabilities, setting a positive trigger position means to capture extra samples that are placed between the sample that meets the trigger condition and the samples that we are interested in. Simply put, if we set the trigger position to 100ms we are actually expecting a buffer of data that happened 100ms after the trigger condition has been met. The extra samples mentioned earlier correspond to the delay we want (100ms for example).
The problem is that delays can be very large (e.g. capturing at 100MSPS, time base = 1 microsecond/div and trigger position = 2s will require a buffer size around 2* 10^8 samples). Both libiio and GNURadio limit the size of buffers that Scopy can request. Additionally why transfer a huge amount of samples just to drop them afterwards since their only use was as a delay.
Currently we limit, in software, the number of extra samples to 500000 samples which limits the positive position that can be set for the trigger.

acostina, 2017-06-16:
Added new registers, renamed the old trigger offset registers to fifo_depth. Simulated the adc_trigger and it works correctly. Added the same mechanism to axi_logic_analyzer. Register map documentation available on wiki at https://wiki.analog.com/resources/fpga/docs/axi_adc_trigger#register_map and https://wiki.analog.com/resources/fpga/docs/axi_logic_analyzer#register_map

lclausen-adi, 2017-06-16:
Driver support: https://github.com/analogdevicesinc/linux/commit/fb2c42996096ac7cb15e9f6f39258e0191387c47

adisuciu, 2018-06-12:
We can implement this in a number of ways:

We can have another spinbox which would be called acquisition delay(or something like that) that would activate this. Basically offset the whole acquisition by the number of seconds specified in the spinbox
Advantage: Easily control how much data you are getting
Disadvantage: Another UI control, confusion on having 2 UI controls that seem to do the same thing (for the unexperienced, Position and Delay)

We can have a checkbox which would activate the acquisition delay. When moving your trigger to the left (increase amount of data before the trigger) exceeding the number of samples visible on the plot, the software starts adding in the acquisition delay. When the checkbox is not activated, the number of samples can be increased to the threshold Dan mentioned (500000 samples).
Advantage: Only one UI control for position
Disadvantage: UI control, that is kind of abstract (in my opinion) - only activate delayed acquisition after the soft threshold is reached

We can have no  UI controls. Moving the position past the hard threshold automatically enables the acquisition delay. Basically you would always get as much data as possible, and when it is no longer possible to get data, you start putting in the acquisiton delay.
Advantage: Ease of use
Disadvantage: You might get too much data which you are not interested in. It can be painful when exporting, analysing, etc. There is no way to control the amount of data. Might as well lower the threshold.

dNechita, 2018-06-12:
I think that the implementation specifics should not be visible to the user.  The instrument could decide which approach to use and when.

rgetz, 2018-06-12:
Yes - lets emulate instruments. 

Instruments don't always expose all the knobs to end users. End users just want to move the trigger position and get the results. Mihai could comment more as well.

adisuciu, 2018-07-27:
Implemented with https://github.com/analogdevicesinc/scopy/pull/452

amiclaus, 2018-08-02:
Checked on on Scopy version: 01d69df (v1.0.1) and the feature is implemented.

closed 2018-08-02 by amiclaus