analogdevicesinc/scopy #51: square waves are not square

rgetz, 2017-03-13:
On Alpha-2:

To reproduce:

Set function generator to 1V, 1kHz square wave.

Connect Channel 1- to GND, and channel 1+ to function generator. The measurements are correct - it's not 50% duty cycle.

[image]

pcercuei, 2017-03-14:
This only happens when using oversampling in HDL.

rgetz, 2017-03-14:
So, this is a bug in the HDL? Or the software needs to drive things differently when the oversampling HDL is enabled?

lclausen-adi, 2017-03-14:
Any duty cycle imbalances will be magnified by the interpolation block by the interpolation rate. Lets say there is a one more low than high sample at the input of the interpolator and the interpolation rate is 10000, than the output waveform will have 10000 more low samples than high samples. This can cause a visible difference.

rgetz, 2017-03-14:
So, this is a question of what is more important, frequency accuracy, or "shape" (in this case duty cycle) accuracy?

lclausen-adi, 2017-03-14:
It is a resolution issue. The lower the oversampling rate (and thus the higher the actual data rate) the better your per sample resolution gets. The relative error you get in both duty cycle and period correlates with the ratio of resolution and sample rate of the signal.

So we might need to rethink the heuristic that is used to choose the oversampling rate to ensure that the error is below a certain threshold (in %).

rgetz, 2017-03-14:
Or provide more limitations on the GUI. (When doing square wave, read back the actual frequency, 1.01kHz (or whatever), and indicate to the user that it being used. It's only confusing since the GUI displays an exact 50% duty cycle 1kHz square wave. 

Being within +/- 2% should be acceptable.

commit c330fcd3 (2017-03-15): Signal Generator: Enforce (rate / freq) ratio to be even for squares

To generate a period of a square waveform we need an even number of
samples.

Fixes #51.
closed 2017-03-15 by pcercuei