analogdevicesinc/scopy #53: running function generator at same time as scope, causes crashes

CElkhoury, 2017-03-13:
This is all I get: 

Unit it plugged into 5.5V AC adapter, so it is not a power issue.

Running the 0.1 firmware, and the 0.2 scopy.

open the oscilloscope, and click "run",
goto to the function generator, and click "run"

get this:

[image]

If there is any way to get more debug info, please let me know.
-Robin

mhennerich, 2017-03-13:
0.1 had the limited USB endpoint issue.
You need to use the latest firmware.

-Michael

rgetz, 2017-03-15:
Which is where? 
https://github.com/analogdevicesinc/m2k-fw/releases

closed 2017-03-15 by rgetz

mhennerich, 2017-03-15:
check your email

mhennerich, 2017-03-15:
created a new v0.11 release

-Michael