analogdevicesinc/scopy #55: AWG Buffer file format?

damercer, 2017-03-24:
What format are you expecting the Buffer file to be in? I tried loading a .csv file ( one column of plain ascii text ) with 4096 rows. The contents was -1, -0.5, 0, 0.5, 1. The reported length was 4301 samples? The analog output was a constant 0 V. I then tried a plain .txt file with one number per line, 4096 lines, contents was -2047, -1023, 0 , 1023, 2047. Figuring maybe the numbers needed to be binary 12 bits. The reported length was 6761 samples? And again the analog output was ~0 V.

Plain text .csv files from EXCEL would be the preferred format.

Could the AWG buffer file feature be further documented?

Thanks

Doug

damercer, 2017-03-24:
[5_steps.zip](https://github.com/analogdevicesinc/scopy/files/868881/5_steps.zip)

pcercuei, 2017-03-27:
Right now the buffer is supposed to contain packed (binary) 32-bit float samples. I've been working on adding support for Matlab's .mat files, and I plan to add CSV later, but it's still not implemented. I'll work on it once I'm done with the network analyzer.

damercer, 2017-03-27:
Not to sound too critical, but what software would a user be using to generate a binary file? Excel comes to mind as a first choice and thus ascii csv. Audio wav files would a second option and they are 16 bit integers? Other AWG waveform editors that are freely available from bench instrument vendors also output csv. 

The reason I'm asking for a way to import a waveform is that I'm looking to reproduce the transistor I/V curve tracer Lab in Scopy. I need a stair step to drive the base synced to the ramp for the collector voltage.

Thanks

Doug

rgetz, 2017-03-27:
Doug:

The goal is either:
-  .txt (text, tab or space delimited) 
- .csv (comma separated variable) 
- .mat (MATLAB) files
- .bin (binary floats)

format would be either:
- floating point (to arbitrary scale)
- integer (to arbitrary scale)

Either can be scaled and offset by the GUI after the file import.

Is that what you are looking for?

damercer, 2017-03-27:
> Doug:
>
> The goal is either:
When you say "either" that can be confusing, you actually mean support 
for all of the above? I would think different users might be generating 
waveforms in lots of different software. Even audio .wav files possibly? 
Some professors like to have their freshman classes listen to their 
favorite audio files when played through different RLC networks.
>
>   * .txt (text, tab or space delimited)
>   * .csv (comma separated variable)
>   * .mat (MATLAB) files
>   * .bin (binary floats)
>
> format would be either:
>
>   * floating point (to arbitrary scale)
>   * integer (to arbitrary scale)
Again Both? In an ascii file integer text is just a subset of numbers 
that have decimal points.
>
> Either can be scaled and offset by the GUI after the file import.
Yes that would be great and I might request that frequency ( or sample 
rate ) also be scalable. Waveforms lets the user adjust amplitude, 
offset, phase ( delay ) and frequency after the file is loaded.
>
> Is that what you are looking for?
Yes, but right now a simple .csv list of samples ranging from -5.00 to 
+5.00 will go a very long way to covering most use cases.

adisuciu, 2018-01-12:
32b366c19ba95b94c751f8eac6f4f069745f7e98 and 5357a6ed74fc298dd0d3b412d4f2627f8f88022c adds support for WAV, CSV and MAT files.

amiclaus, 2018-01-16:
On Scopy build e9a6c85 (v0.10) when I try to import a WAV file (16bit), i.e. [](https://freewavesamples.com/), Scopy crashes.

adisuciu, 2018-01-16:
5a1ce862edd00e3b45329f29a62d9277ce596d24 should fix this.

amiclaus, 2018-01-18:
On Scopy build: d85de1c (v0.10) I've tested the importing feature with the following files (.csv, .wav, .mat):

[Downloads.zip](https://github.com/analogdevicesinc/scopy/files/1642307/Downloads.zip)

and it works.

Therefore, I will close the issue.

closed 2018-01-18 by amiclaus