analogdevicesinc/scopy #56: Voltmeter not actually reading voltage applied at input

damercer, 2017-03-26:
Hi:

Here is what I'm doing. Connect the variable positive supply to the channel 1 + input. Ground the channel 1 - input. Connect a DMM to the same place. Plug in the M2K. Start Scopy. Open the power supply instrument. Set the voltages to +2 and -1, enable both supplies and hit the run button. DMM verifies the +2 Volts on the input. Open the voltmeter instrument. Hit run. The display shows 0.114 V ( with or with out the power supply turned on ). You can change the voltage and the voltmeter always reads the same. The same happens on channel 2.

Open the scope instrument and hit run. Again the channel 1 trace is at about 100 mV not the +2 V that is applied at the input.

It looks like there are some switches that actually connect the inputs to the ADC are not being turned on in the right way.

Running the latest alpha release, Win 7 Pro 64 bits.

Doug

damercer, 2017-03-26:
OK, It would seem that one of these two M2K's that I have is broken. Probably from day one. Maybe the firmware is corrupted? How do I get this non-functional M2K to work properly?

Thanks

Doug

pcercuei, 2017-03-27:
I think upgrading to the latest firmware should give you a functional state, if it doesn't then it's probably toasted...

damercer, 2017-03-27:
And how does one "upgrade to the latest firmware"? Remember I'm not a developer just a dumb user so assume nothing.

rgetz, 2017-03-27:
Doug:

The instructions are the same as:
https://wiki.analog.com/university/tools/pluto/users/firmware

But use the M2k firmware, not the Pluto one:
https://github.com/analogdevicesinc/m2k-fw/releases

This is why I asked if the mass storage device was showing up.

However - all the devices I passed out had 0.11 on them - so I don't think that is the issue.

damercer, 2017-03-27:
Reloading the firmware fixed the non-functioning M2K.
Closing this issue.
Doug

closed 2017-03-27 by damercer