analogdevicesinc/scopy #57: AWG output phase between channels seems random

damercer, 2017-03-27:
The relative phase between channel 1 and channel 2 seems to be random, different each time stop, run, stop, run .... Both channels are set to same settings, ampl, offset, freq, phase, shape. Both signals should lie on top of each other.

The reason I'm concerned about this is I'm trying to reproduce the transistor I/V curve tracer Lab and I need the base ( stair step ) and collector (ramp)  waveforms to be properly aligned at all times.

Doug

pcercuei, 2017-03-27:
Yes, I can reproduce. This happens on the integrated version, not on the FMC version.
The 'DMA sync' feature seems to be missing in the HDL.

lclausen-adi, 2017-04-03:
My fault. Fixed with https://github.com/analogdevicesinc/linux/commit/35653174ab0f7242cd1cd815951fd3ea80640a31

closed 2017-04-03 by lclausen-adi