analogdevicesinc/scopy #58: Connecting/Disconnecting during calibration stops the application.

amiclaus, 2017-03-29:
Connect M2K to the Desktop/Laptop. Start Scopy, select the detected device. Press "Connect" button. After the connection is made, press again the button (before calibration is done). 
Results: Scopy application stops responding. Message box appears indicating that the "Scopy has stopped working". Applications is forced to close.

pcercuei, 2017-03-29:
This happens because the IIO context is destroyed while the calibration is running. The proper fix would be to have a way to abort the calibration.

rgetz, 2017-03-29:
Or just ignore connect/disconnect events while calibrating? 

Is it possible the device hangs during calibration? That would be the only reason to not block things, and go through the effort of calibration cancellation.

adisuciu, 2017-11-06:
Fixed with 7c5e247

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus