analogdevicesinc/scopy #59: Disconnect Button

amiclaus, 2017-03-29:
Connect M2K. Start Scopy, select the detected device. Press "Connect" button. 
Actual Results: The button color changes to green and text changes to "Connected", without any hint how to disconnect the board.
Expected Results: When connection is made, an additional button should appear near the "Connected" button. This button should have the text "Disconnect" or "X" and when it is pressed, the board will disconnect. The status of the connection can also be separated from the buttons (text/color of buttons don't change, status of the connection is displayed somewhere else on the interface).

acozma, 2018-01-17:
This is useful in the context of the device manager.

adisuciu, 2018-03-07:
Connect button label was changed with commit 4c5cc38c17f29569035226dd48e5bed96596daa6

amiclaus, 2018-03-12:
On Scopy build: 6bd7075 (v0.11) the feature is implemented.

closed 2018-03-12 by amiclaus