analogdevicesinc/scopy #60: Scopy Minimum Size window make texts and buttons overlap.

amiclaus, 2017-03-29:
[image]
[image]
[image]

Connect M2K. Start Scopy, select the detected device. Press ""Connect"" button. Select the Window to be displayed in its minimum size.
-select Oscilloscope; press Channel Settings button; observe the dropdown menus for selecting the proper measurement units (Ex. Vertical Volts/div ->V;mV;µV) ;
-select Oscilloscope; press Trigger Settings button; 
observe the overlapping of the External Option (on/off) overlapping the Trigger Mode Option.
-select Oscilloscope; activate the Cursors (make them visible on plot); press FFT Option button (wheel icon on top right of the window); check the FFT feature; observe the overlappings (between cursor values; channel 2 vertical display scale is covered by the plot);
-select Pattern Generation; observe the values of the plot time axis overlapping;
Actual Results: Texts, values or buttons are overlapping in minimum sized Scopy window.
Expected Results: Texts, values or buttons should be resized according to the Scopy window size.

rgetz, 2017-03-29:
I think the design documentation states the min screen size is 1024x768. 

What size/resolution is shown above?

pcercuei, 2017-03-29:
@rgetz: we decided for 720p - 1280x720.

amiclaus, 2017-03-29:
@rgetz : the window size shown above is 1296 x 758 on a screen resolution of 1600 x 900.

dNechita, 2017-05-25:
@rgetz There is no mention in the documentation regarding the minimum screen size.
A reasonable resolution that me and Paul thought of is 1280x720.

pcercuei, 2017-05-25:
The major problem we have is that all the GUI mockups are designed for 1080p and above... and don't work as good at lower resolutions.

rgetz, 2017-05-25:
So, this is a mockup issue, that should be assigned to Andrei? :)

I think we need to support at least down to 720p. There are still laptops shipped with:
 - 1280×720 px (HD)
 - 1280 x 1024 (SXGA)

pcercuei, 2017-05-25:
The MacBooks are 2560x1440, with a Hi-DPI display, which means that the effective resolution for the application is 1280x720. So 720p is definitely a necessity.

I think we can skip SXGA, though.

amiclaus, 2017-05-26:
I'll close this post since I've splitted it into #114 , #115 and #116

closed 2017-05-26 by amiclaus