analogdevicesinc/scopy #61: Independent Run option for each channel

amiclaus, 2017-03-29:
Connect M2K. Start Scopy, select the detected device. Press "Connect" button. After the connection is made, select Voltmeter feature. Press Run button.
Actual Results: Both Channels are running.
Expected Results: Enable/Disable Channel feature should be introduced for each channel so that they can run independently (similar to the Power Supply Feature).

pcercuei, 2017-03-29:
That's debatable. I don't see any problem in having the two channels running at the same time.

rgetz, 2017-03-29:
Yeah, I'm not sure why someone would need to turn on a channel on a voltmeter. That's not something you do on a fluke - a software defined instrument should be the same.

amiclaus, 2017-05-26:
Since, there are no other opinions and taking into consideration the feedback received, I will close this post.

closed 2017-05-26 by amiclaus