analogdevicesinc/scopy #62: History Reverse Axis Option

amiclaus, 2017-03-29:
"Connect M2K. Start Scopy, select the detected device. Press "Connect" button. After the connection is made, select Voltmeter feature.
Actual Results: History Time Axis -> Vertical; History Voltage Axis -> Horizontal. 
Expected Results: Reverse Axis Option should be introduced so that the user can switch axis: History Time Axis -> Horizontal; History Voltage Axis -> Vertical. "

rgetz, 2017-03-29:
Can you describe a student centric lab, where this feature is necessary? 

I'm not sure I understand the purpose. The stripchart in the voltmeter tool is a stripchart. 
https://en.m.wikipedia.org/wiki/Chart_recorder

All voltmeter/stripchart tools that I am familiar with use a galvanometer, and things flow down. 

If you need time to flow horizontally, that's a oscilloscope.

amiclaus, 2017-03-29:
This feature is not necessary, but I find it useful since when I first used the Voltmeter feature I was a bit confused about that representation and its meaning, being accustomed with the evolution of a signal in time being described using the horizontal axis for the time variable.

rgetz, 2017-03-29:
That's because you are already familiar with a scope. :)

damercer, 2017-03-31:
There are some physical strip-chart like devices that have time on the horizontal axis. Like a lie detector for example. Like most things one can always find exceptions to the "rule". Just because of a preconception of what is proper should not automatically rule out other options. Most computer screens ( tablets as well ) are used in landscape orientation so the long axis ( which in most cases is time ) will be horizontal.

amiclaus, 2017-05-26:
Since there is no clearly stated favorable feedback regarding this enhancement, the post will be closed.

closed 2017-05-26 by amiclaus