analogdevicesinc/scopy #63: Reposition Measured Voltage Display

amiclaus, 2017-03-29:
Connect M2K. Start Scopy, select the detected device. Press "Connect" button. After the connection is made, select Voltmeter feature. Set the History Button to Off. The Horizontal Voltage Axis resizes.
Actual Results: Measured Voltage Display position remains unchanged.
Expected Results:  Measured Voltage Display position should change according to the new resized Horizontal Voltage Axis. (move to a more central position).

rgetz, 2017-03-29:
I'm not sure what this is asking for. Can you include a screen shot?

amiclaus, 2017-03-29:
[image]
[image]

Since the horizontal axis on which the measured voltage is displayed is resizing, the numerical voltage display should adjust its position according to the centre of the axis.

rgetz, 2017-03-29:
I'm not sure why it would. 

That's not the way a Fluke works. 
http://en-us.fluke.com/products/digital-multimeters/fluke-87v-digital-multimeter.html

The display (7-segment display) is fixed on the LCD.

If you think there needs to be more separation between the bar, and the reading - that's different, but I don't think the text should change.

amiclaus, 2017-03-29:
According to Fluke, the horizontal bar is also fixed. So maybe the axis size shouldn't change based on the fact that the history is on/off.

lclausen-adi, 2017-03-29:
Is there any good reason to disable the history? If not maybe just keep it always enabled and remove the enable/disable button.

rgetz, 2017-03-29:
There needs to be some way to autoscale/reset things - so I can see the value in clearing history. 

Let me read the fluke Manual later.

damercer, 2017-03-31:
I think that when the History ( strip chart ) is enabled the bar graph is taking a disproportionate fraction of the available space. I would give more space to the strip chart when enabled, i.e. more pixels means more visual resolution on the voltage axis.

pcercuei, 2017-04-03:
I think the strip chart should resize when increasing the window size; right now it looks good at 720p, but really disproportionate at higher resolutions.

AlexandraTrifan, 2018-05-24:
The issue regarding strip chart resizing should be fixed after commit d3986f0cfd518ab5dd02ae8a5ae85164877a921e.  
Do we want to move the digital display when the History is disabled?

closed 2020-10-20 by adisuciu