analogdevicesinc/scopy #64: Distinguish between the same axis cursors.

amiclaus, 2017-03-29:
Connect M2K. Start Scopy, select the detected device. Press "Connect" button. After the connection is made, select Oscilloscope feature. Check the Cursors to be enabled.
Actual Results: Each pair of the horizontal/vertical cursors are identical.
Expected Results: The cursors should be distinguished through suggestive color/text in such a manner that a correlation will exist between the cursors value box and actual cursors.
[image]

pcercuei, 2017-03-29:
The handles could have the '+' and '-' signs, for instance.

rgetz, 2017-03-29:
I think "A" and "B" is pretty common. 

("+" and "-" only make sense on the voltage scale). 

However - no scope that I know of actually indicates this on the screen. (Labels the cursors). 

https://cdn-shop.adafruit.com/datasheets/Users+Guide+DS1000E.pdf

Page 2-96 

The active cursor is highlighted/shaded differently.

dNechita, 2017-07-25:
In this case it can be said that all 4 cursors are active since there's no button that selects between cursors and there's no need for one.
What about the suggestion presented in the image below?
[image]

rgetz, 2017-07-25:
I think that clutters up the screen.

I don't think you need labels. The delta is always positive, it doesn't matter which is which.

dNechita, 2017-07-26:
Yes, I would also prefer to keep things simple. Things have already started to get crowded at the edges of the plot (2 handles for cursors, handles for trigger level, for offsets, for time position, and all sorts of labels and possibly more to come).

One can easily identify which cursor is which by simply moving around one of the cursor while keeping an eye on which cursor readout is changing values.

Therefore I'm closing this issue.

closed 2017-07-26 by dNechita

rgetz, 2017-07-26:
Sounds great.