analogdevicesinc/scopy #65: Add "Track Waveform" Feature

amiclaus, 2017-03-29:
Connect M2K. Start Scopy, select the detected device. Press "Connect" button. After the connection is made, select Oscilloscope feature.
A "Track Waveform" feature should be added, that manages the horizontal cursor lines to follow the min./max. values of an acquired signal,  the vertical cursor lines to mark off a period of a signal.
Such a feature is presented in the attachments.
[image]
[image]

rgetz, 2017-03-29:
Can you describe a student centric lab where this feature is necessary? Rather that just turning on a p-p measurement and reading it?

amiclaus, 2017-03-29:
This is not a necessary feature, but can improve the understanding of what a peak-to-peak measurement value really represents from the graphical point of view. Also, such a feature provides better accuracy in cursor positioning, rather than manually place them.

rgetz, 2017-03-29:
There are real bugs to fix/improvements to make. 

Things like this are valuable, but right now - would go on the bottom of the list to do.