analogdevicesinc/scopy #66: Inappropriate initial channel settings panel title display

amiclaus, 2017-03-29:
Connect M2K. Start Scopy, select the detected device. Press "Connect" button. After the connection is made, select Signal Generator feature. The channel settings panel is automatically opened for channel 1.
Actual Results: The title of the panel initially is "Configuration", not specifying for which of the channels the panel is active.
Expected Results: The title should be set to "Configuration for Channel 1" when opening the Signal Generator feature for the first time.
[image]

adisuciu, 2017-07-24:
Fixed with 	5555389

amiclaus, 2017-07-25:
On Scopy build: f759590 (v0.6) the bug is fixed.

closed 2017-07-25 by amiclaus