analogdevicesinc/scopy #71: Group With Selected generates the crash - Missing Decoders

amiclaus, 2017-03-29:
Connect M2K. Run Scopy via build folder generated through MSYS/Qt, select the detected device. Press "Connect" button. After the connection is made, select Pattern Generator feature. Select multiple DIO and press the button Group with Selected. 
Actual Results: The Scopy Application stops working. This issue happens due to the fact that the decoders folder is missing from the build folder.
Expected Results: Application should continue working properly and decoders folder should be somehow accessed or copied to the build folder.

pcercuei, 2017-03-29:
I don't think this is a bug, actually, as it won't happen in the release of Scopy installed with the setup installer, but only on a local development build.

dNechita, 2017-03-29:
Some users might want to build Scopy from scratch and they will have this problem and won't know why crashes occur.
We should add documentation for this.

rgetz, 2017-03-29:
Is this because the "correct" git submodules aren't set up properly?

pcercuei, 2017-03-30:
There are no submodules. The 'decoders' are provided by libsigrokdecode and can be found in /usr/share/libsigrokdecode/decoders on Linux, or C:\Program Files\Scopy\decoders on Windows.

The bug here is when trying to run a dev build of Scopy if those decoders have not been installed.

adisuciu, 2017-04-13:
I implemented c7480f639c845ca5662d2b687a13f2e15eb6e42f which shows an error message when libsigrokdecode or decoders cannot be initialized properly. Also the application no longer crashes when this happens.

closed 2017-05-03 by adisuciu