analogdevicesinc/scopy #72: Drop down menu partially not working

amiclaus, 2017-03-30:
Select "Measure" Feature from the bottom right side of the Oscilloscope window. In the section "Custom Selection" press the Horizontal/Vertical drop down menu button. 

[image]

Actual results: the drop down list appears only if the right side of the button is selected (the one containing the down arrow indicator).
Expected Results: the drop down list should be available by pressing the button anywhere within its region.

DanielGuramulta, 2017-08-24:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/18190b64ff458e49fb4d8ab24959a8b6b9182a86

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus