analogdevicesinc/scopy #75: Logic Analyzer - Memory Leak

amiclaus, 2017-04-03:
Connect M2K. Start Scopy and connect to the detected device. Select the Logic Analyzer feature. Run the Logic Analyzer. Open Task Manager, select the Processes tab and observe the Scopy.exe process.
Actual Result: The memory used by the process is increasing continuously.
Expected Result: There should be no memory leak while running the Logic Analyzer.

[image]

AlexandraTrifan, 2017-05-05:
Commit 720f4cd915823c9f0c9df94ee3737669e3faefeb introduces some fixes that reduce the memory usage.  I ran some tests and the memory used did not grow so much in a short period of time. This problem is not completely solved yet, but the tool is more stable now.

AlexandraTrifan, 2020-09-25:
This no longer happens using the latest version and the new Logic Analyzer. https://github.com/analogdevicesinc/scopy/releases/tag/v1.2.0-rc.1

tagoylo, 2020-10-21:
This issue does not happen on v1.2. Closing this item.

closed 2020-10-21 by tagoylo