analogdevicesinc/scopy #76: Streaming issues with ADC at 1 kSPS / 10 kSPS

pcercuei, 2017-04-03:
On both the M2K integrated version and FMC version, setting the ADC sampling rate to 1 kSPS or 10 kSPS will rapidly result in stalls.

This can be verified by running `iio_readdev m2k-adc | hexdump` on the M2K.

lclausen-adi, 2017-04-03:
What do you mean by stalls? DMA completely freezes?

pcercuei, 2017-04-03:
It looks like so; yes.

acostina, 2017-04-13:
I've tried to get data directly accessing the drivers through the file systems and I didn't encounter a stall.

Steps performed (directly through the terminal, on a standalone scopy, with Signal Generator on  at 5 HZ sines on both channels ):
cd /sys/bus/iio/devices/iio:device3
echo 1000 >sampling_frequency
cd buffer
echo 16 > length
echo 1 > enable
od -s /dev/iio:device3 -w4 -v

Also, written 100 to oversampling_ratio and new data is displayed every 1-3 seconds.
I think the issue is somewhere in the software layers, not HDL.

pcercuei, 2017-04-25:
I can't reproduce it anymore now, so I'll close the bug.

closed 2017-04-25 by pcercuei