analogdevicesinc/scopy #77: Unstable parameter computations when applying ramp/triangle signal.

amiclaus, 2017-04-04:
Connect M2K. Start Scopy and connect to the detected device. Open the Oscilloscope and enable the Measure feature. 
Apply a ramp/triangle signal with any frequency 1Hz-200kHz (5Vpp). 
Observe the measurement box values.
Some of the parameter values: Low, High, Amplitude, Middle, +Over, -Over, Rise, Fall, +Width, -Width, +Duty, -Duty are fluctuating too much due to the fact that the some of these parameters cannot be computed properly.
Should we somehow disable the display of these inaccurate values in these specific situations?

claudiupop, 2018-07-17:
The problem with the measurements on triangle / sawtooth signal is that the histogram generated is random (See picture) when it should be uniform. To calculate low / high value it picks the  most common value found above / below midpoint (on histogram). Because the most common value is changing often, all the measurements, that use low / high value for computation, are fluctuating.

[image]

We connected the m2k to an oscilloscope. For the same signal, the scopy failed to calculate the low / high values, while the oscilloscope displayed a constant value for low / high. Eg: for a triangle wave at 100 kHz 5Vpp the oscilloscope measured:
Min.Value: -2.594v 
Max.Value: 2.531v 
Low: -2.484v 
High: 2.432 

while scopy measured :
Min.Value: -2.534v 
Max.Value: 2.550v and low/high were fluctuating.

On oscilloscope the values for low/high were very close to min/max values but different.

dNechita, 2018-07-19:
There is no predominant component in a histogram for signals like triangle and sawtooth. If you can detect that situation you should fall back to other methods of detecting 'high' and 'low' measurements. The histogram approach is just not that effective for these types of signals. 
No ideal methods come to my mind but some ideas I have are:
- for detecting the 'low' get a couple of the lowest values and since you now know their distribution, you could compute a weighted arithmetic mean for those values.
- or apply a low pass filtering and then try again the histogram approach

adisuciu, 2020-07-16:
Most scopes don't have this functionality. One should understand the way measurements work. I'm just gonna close it.
-Adrian

rgetz, 2020-07-16:
Most scopes do have this functionality.

It's a simple way - keep track of successive measurements (across buffers) - if they vary more than 10% - display "-----" rather than the number.