analogdevicesinc/scopy #79: Innacurate Amplitude value and distorted signal at high frequencies

amiclaus, 2017-04-04:
Connect M2K. Start Scopy and connect to the detected device. Open the Oscilloscope and enable the Measure feature. 
Apply a rectangular/sinusoidal signal with frequency 10MHz, 5Vpp.
Observe the measurement box values.
In case of a rectangular/sinusoidal signal, the Amplitude is inaccurate (over 7V/6V p-p).
In case of a rectangular waveform, the signal is considerably distorted. 
Maybe we should provide a max signal frequency value in the documentation at which the nr. of samples are enough to visualize properly the signal.

rgetz, 2017-04-04:
Where is the 10MHz, 5Vp-p waveform coming from? The M2k? Or a function generator?

amiclaus, 2017-04-04:
I

amiclaus, 2017-04-04:
I've remade the measurements and the Amplitude of the signal is ok, either using the waveform from M2K or from a function generator.

closed 2017-04-04 by amiclaus