analogdevicesinc/scopy #82: Oscilloscope needs to be able to plot samples as soon as they are available

dNechita, 2017-04-10:
About:
Currently the oscilloscope tool waits for the entire data to be captured then displays it. This works well for small time bases like a couple of microseconds/division up to maybe 1-2 milliseconds. But for higher time bases the frame rate will be severely affected and can even cause waiting for a period of time until a frame is actually plotted.

One possible solution:
There should be a threshold (a certain value of the time base) that would make the tool switch from this 'buffer' mode to a 'streaming' mode where samples are being displayed as soon as they are available.

AlexandraTrifan, 2018-03-07:
This was implemented with commit 8292c11aff01fda059ccf8bf3bd1ddd991ee67b0. 
The oscilloscope tool automatically switches to a different mode, where the samples are not displayed all at once, but sequentially. The threshold for this switch is a timebase of 100 ms/div.

amiclaus, 2018-03-12:
On Scopy build: 6bd7075 (v0.11) the feature is implemented.

closed 2018-03-12 by amiclaus