analogdevicesinc/scopy #83: Auto Scale the input when use is scaling the amplitude range.

CElkhoury, 2017-04-12:
The high vs low gain needs to automatically update based on the user and input signal.

Currently it's always on high gain.
[image]

seems as if the lsb is always 16mV. should be less than 1mV.

Charly

damercer, 2017-04-13:
I think this is more or less the same question asked in issue number 73.

Doug

rgetz, 2017-04-21:
this issue is blocking people from looking at the input channels.

Can we get a schedule of when the nightly build might fix this? or a temporary by hand workaround? (to flip to the other input range?)

-Robin

dNechita, 2017-04-24:
If this switch between gains is to be done automatically by the instrument then what is the factor that determines the switch to take place?
The only thing I can think of is the Volts/Div going under/over a certain threshold. If so, how to determine the right threshold?
Are there any other factors?

damercer, 2017-04-24:
Yes, generally the gain range is switched based on the V/Div setting. The position is handled by the addition of the DC offset in the hardware. When to switch is a function of the Max/Min voltage in each gain range. The gain is switched when the screen height is just less ( next V/Div setting would show signal clipped when ADC over ranges ) than the Max/Min voltage span.

I think this is how Digilent does it in Waveforms.

Doug

damercer, 2017-04-24:
I have a general design question with regard to the analog input voltage divider gain. What is the intended Max supported input voltage? I just tested where the positive voltage clips and it seems to be about +32.5 V. What are we going to specify and if it is going to be +/-25 why not change the divider ratio to get a smaller LSB in the high range?

Analog Discovery is speced at +/- 25 but I just tested AD2 and it does not clip until above 35 V? ( 14 bit ADC vs 12 bit ADC so they can afford higher range ).

Doug

rgetz, 2017-04-24:
https://reference.digilentinc.com/reference/instrumentation/analog-discovery/reference-manual#scope_input_divider_and_gain_selection

i think that describes the input section. It is less than 50V (differential) and less than 7V. (differential). As you noted - it is specked at  ±25V; since there is a safety thing you need to meet if you advertise more than that - which we did not want to do (which I think is a UL thing, but I can't find it at the moment).

I think we wanted to be functional equivalent so we used the schematics that we gave Digilent when we designed the analog front end for them.

damercer, 2017-04-24:
I think +/- 25V as the max input range seems right. The use of a 10 X scope probe provides additional voltage range as needed. The question is how much bigger does the clipping level need to be than the stated range. Right now we are at 130 % ( 65/50 ). I think 110% should be enough to allow for calibration.

dNechita, 2017-05-28:
Implemented in commit https://github.com/analogdevicesinc/scopy/commit/23a0c7710798ac7a8132f1cf94248f710c21ab33

amiclaus, 2017-07-06:
On Scopy build: 21bb304 the bug regarding the automatic switch between high gain and low gain is fixed. Moved the discussion for the maximum supported input voltage to #146

closed 2017-07-06 by amiclaus