analogdevicesinc/scopy #85: M2K - Scopy calibration

damercer, 2017-04-15:
I've noticed something about how the M2K gets calibrated by Scopy. The first time Scopy is run after an M2K boots up ( freshly turned on ) the AWG output calibration is way off. Terminating Scopy and re-running it the calibration is about right ( but drifts off a lot as the board heats up ). Here are some M1K scope plots showing the issue with my two M2K boards. The signal generator is set to 1 KHz sine wave with offset to 2.5 and amplitude set to 2.0.
[image]
[image]
Closing and re-running Scopy we get these two results:
[image]
[image]
Here is a plot of the M1K connected to an AD584 2.5 V reference for checking purposes.
[image]
I then set the M2K signal generators to DC and 2.5 V to check for signal accuracy and quality.
After significant warm-up time board no. 1 shows a negative drift of the DC value and randomly spaced jumps of a bout 10 mV.
[image]
Board no. 2 shows about 20 mV of a fixed pattern noise repeating at about 400 uS.
[image]

So this is to report on what appears to be a software calibration issue as well as a potential Board design issue ( perhaps related to the analog power supplies?)

Id there a way to force a recalibration while Scopy is running?

Doug

mbanciso, 2017-04-19:
I looked into this and the DAC calibration fails on the first try because after writing to the _DAC Gain Ctrl 0_ register, we need to wait for about 15 seconds until the output settles.
In order to make this delay less visible to the user, I think we need to: 
1 - write these registers right after linux booted:
        _iio_device_reg_write(dev, 0x68, 0x1B);  // IGAIN1 +-6db  0.25db steps
        iio_device_reg_write(dev, 0x6B, 0x1B);  //
        iio_device_reg_write(dev, 0x69, 0x1C);  // IGAIN2 +-2.5%
        iio_device_reg_write(dev, 0x6C, 0x1C);
        iio_device_reg_write(dev, 0x6A, 0x20);  // IRSET +-20%
        iio_device_reg_write(dev, 0x6D, 0x20);_
2-  Have a flag which gets cleared after ~15 sec - so Scopy knows when it's OK to perform the calibration

lclausen-adi, 2017-04-20:
Why does it take 15 seconds for the output to settle?

rgetz, 2017-04-20:
Sounds like hardware issue.....

rgetz, 2017-04-21:
I had a brief discussion with the chip designer - there is no "output settling" or internal calibration inside the DAC. This is either a circuit (PCB) issue, or a software one.

closed 2024-09-24 by adisuciu